WBK-Robotics/pybullet_industrial #41: [JOSS REVIEW] Software Feedback

sea-bass, 2023-03-05:
Review thread at : https://github.com/openjournals/joss-reviews/issues/5174

Thanks for creating this awesome tool! I went through the source installation process so I could poke around the code, but I enjoy that you also have a pip install possible.

My overall experience was great. Love that there is CI, ReadTheDocs documentation, installation was easy, and I could run your examples very quickly. This is a nice tool and my comments on the software are fairly minor.

# Software issues
* When installing, namely `pip install src/` I got an error that was resolved by first doing `pip install wheel`. Might be worth either saying somewhere, or including it in your requirements file
* The reference code in the [Generating paths tutorial](https://pybullet-industrial.readthedocs.io/en/latest/tutorials.html#generating-paths) crashes for me with this error.
    ```
    b3Printf: User debug draw failed
    b3Printf: b3Warning[examples/SharedMemory/PhysicsClientSharedMemory.cpp,1286]:
    ```
* The `camera_sensor` example failed on me initially because `matplotlib` was not available in my clean environment. Might want that added to the pip requirements file too.

---

# Documentation
* Would be nice in both the README and the full docs to have an image and/or animation of your tool in action
* The README should link to the ReadTheDocs documentation -- something like "for detailed documentation, go here"
* The installation steps should say which versions of Python and which operating systems are supported
* In the [3d printing and milling tutorial](https://pybullet-industrial.readthedocs.io/en/latest/tutorials.html#d-printing-and-milling), the second code block has a dangling `for _ in range(20):` that is actually part of the next block.
* The full code in the [3d printing and milling tutorial](https://pybullet-industrial.readthedocs.io/en/latest/tutorials.html#d-printing-and-milling) seems to be missing the `target_position` and `extruding` variables? The example itself in the `examples/` folder works perfectly fine, though!

liquidcronos, 2023-03-08:
Hey @sea-bass,
I think I have now included everything you ask for.  Let me know if there is anything else i can do!

sea-bass, 2023-03-11:
@liquidcronos 

Thank you! Two comments here on your changes:

* Rather than checking for matplotlib, or warning use they didn't install matplotlib at example running time, why not just add it to your [setup.py](https://github.com/WBK-Robotics/pybullet_industrial/blob/main/src/setup.py#L14)? I would understand if you wanted to keep the installation lightweight, but pybullet is so massive that adding matplotlib won't make a big difference, haha
* Found a small typo in "Example" [here](https://github.com/WBK-Robotics/pybullet_industrial/blob/dc788a474217193f46ea792daade59e4e6d4fb59/examples/robot_frame_visualization.py#L1)

liquidcronos, 2023-03-12:
Thank you for your suggestions I have adressed these in the newest commit.
Regarding matplotlib you are probably right, but on principle i think we shouldnt install libraries that are not needed for the main functioning of the library since only a small subset of people will use the examples.

sea-bass, 2023-03-12:
Sounds good to me -- thank you for going through this feedback!

closed 2023-03-12 by sea-bass