WBK-Robotics/pybullet_industrial #42: [JOSS REVIEW] Document Examples

CameronDevine, 2023-03-07:
All the examples provided are wonderful; however, there really should be documentation provided to go along with each one. At present, the file names are the only indication of what the examples do. Furthermore, several examples are interactive but there are no instructions about how to interact with them.

openjournals/joss-reviews#5174

CameronDevine, 2023-03-08:
I added a few comments to commit f37defe with some minor edits.

liquidcronos, 2023-03-08:
Thank you for pointing out the spelling mistake!
Regarding the pendulum: the endeffector tool itself can also have joints and links therefore a pendulum shaped endeffector object was loaded and force applied to the tcp corresponding to the tip of the pendulum. I have updated the description to make this more clear. I have also since updated the paper to make it clearer that the endeffector tools can have different shapes and even joints which are loaded from URDF files

CameronDevine, 2023-03-13:
Unfortunately, I still don't feel like it is clear what is happening in the `tcp_force.py` example. Looking through the code I think there is a constant force on the pendulum. Could you add another sentence to the documentation describing what forces are applied to the pendulum?

liquidcronos, 2023-03-13:
Hey @CameronDevine, thanks for the feedback.
I have tried to adresse it in the newest commit. 
Let me know if this explains it better.

CameronDevine, 2023-03-13:
I made a final edit to the `tcp_force.py` example in #46.

CameronDevine, 2023-03-14:
Thank writing this documentation. I believe completing this was important as it will be a common place new users to look when trying to understand the capabilities and use of this package.

closed 2023-03-14 by CameronDevine

liquidcronos, 2023-03-14:
Yeah, definetly, thank you for the feedback, if you have written it yourself its always great to have someone from outside provide feedback as one always has the disadvantage of knowing what one meant :D

commit f37defe3 (2023-03-08): adding docstrings to each example explaining their usages. resolves #42