WBK-Robotics/pybullet_industrial #55: How to use milling to achieve high-precision milling of a large object.

liqingxin, 2024-04-22:
When I used milling spawn_material_block and designed a small particle size, the software got stuck.
I need to conduct a cylindrical milling simulation with an accuracy of 0.2mm. I'm not sure if it can be achieved through PI. 
Also, does it support GPU to improve the rendering efficiency of voxels.

liquidcronos, 2024-04-23:
Hi the milling simulation should definitely not just get stuck. 
Could you provide a minimum code example that leads to the problem as described in the bug issue template?
Then we can help you diagnose/solve the problem.

Regarding GPU support, unfortunately Pybullet Itself does not offer it since the Renderer is only supposed to be used for debugging purposes. Running the simulation without the renderer however greatly improves its speed. If you still need images of the final result for examples you can place a camera endeffector in the simulation an render images at specifc times that way.

liqingxin, 2024-04-23:
Hello , 

here is a demo I made based on the PI example.

Because I want to do some high-precision milling with an accuracy of 0.1mm. So I set the particle size parameter relatively small.
Then there will be a warning, registerGraphicsInstance out of range, 524288
I'm  not sure if this value can be artificially increased.

If the parameters are set smaller ,eg 0.01, a simulation can be obtained. However, from the results, the voxel removal is not continuous, and the workpiece will have many tooth marks. I think my milling parameters were set incorrectly, but I am not sure how to improve them.

Email: ***@***.***
Mobile: 18240131209
Tel: 024-23970592
 
From: Jan Baumgärtner
Date: 2024-04-23 13:52
To: WBK-Robotics/pybullet_industrial
CC: 李庆鑫; Author
Subject: Re: [WBK-Robotics/pybullet_industrial] How to use milling to achieve high-precision milling of a large object. (Issue #55)
Hi the milling simulation should definitely not just get stuck.
Could you provide a minimum code example that leads to the problem as described in the bug issue template?
Then we can help you diagnose/solve the problem.
Regarding GPU support, unfortunately Pybullet Itself does not offer it since the Renderer is only supposed to be used for debugging purposes. Running the simulation without the renderer however greatly improves its speed. If you still need images of the final result for examples you can place a camera endeffector in the simulation an render images at specifc times that way.

liquidcronos, 2024-04-23:
Hi, I dont see a Demo,

what I meant is a code snippet with which to replicate the error.
But the error you describe has to do with the GUI itself.
It cant render objects past a threshold. 
But you can still simulate them without the gui.
Otherwise you can also think about using dexels instead of voxels to reduce the particle size or only spawn the particles where you are actually milling.

I hope this helps, but since this is mainly related to pybullet and not pybullet industrial I will close this issue.